ARMmbed/htrun #23: stlink command is hardcoded for Windows

salkinium, 2015-09-17:
Executing
```mbedhtrun -c stlink -r stlink ...```
on OS X results in this error:
```-p: ST-LINK_CLI.exe: command not found```

This is not surprising, since in both plug-ins, the CLI is [hardcoded to this version](https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests_plugins/module_copy_stlink.py#L34).

For OS X (and Linux) this should be `st-link`.
However [the command format](https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests_plugins/module_copy_stlink.py#L53-L57) needs to be different too:
```st-flash --debug write path/to/image.bin 0x08000000```

Unfortunately `st-flash` currently fails on my machine, since it does not recognise my chip id, so I cannot verify that this command actually works.

PrzemekWirkus, 2015-10-09:
How about PR to fix this?
You can use host OS detection functions I've just added to implement it: https://github.com/ARMmbed/htrun/commit/a52e8aff8cf792879934a41d378896c6a8c60c5d

salkinium, 2015-10-09:
I actually have implemented this for OS X, however, I ran into a lot of trouble with the open-source `st-flash` version, which apparently needs to connect under reset?
I then didn't spend anymore time on it.

PrzemekWirkus, 2016-01-04:
This is addressed here: https://github.com/ARMmbed/htrun/releases/tag/v.0.1.18
Now each plugin can specify if it works under given OS (in our case Stlink thing will not be available under !Windows OS.
You can use ```mbedhtrun  --plugins``` to check if plugin works under your OS or not.

closed 2016-01-04 by PrzemekWirkus