ARMmbed/htrun #25: Serial output is cut off and test times out (on OS X)

salkinium, 2015-09-21:
Running [a test with more output than usual](https://github.com/salkinium/mbed-drivers/blob/feature/spi_unittest/test/asynch_spi/asynch_spi.cpp#L117) results in a cut-off of serial output and the test ultimately times-out.
The serial output buffer is apparently also not properly emptied, as the same partial output appears twice.
https://gist.github.com/salkinium/4d46cb487bbb8a292e46

This does not happen when reducing the serial output of the test. 
It also does not happen when using a non-Python terminal emulation program, like picocom.

There is also buffered output in picocom when opening it on the same port any time after the `yt test` timed out:
https://gist.github.com/salkinium/0391bbd70f9071580d5e

autopulated, 2015-09-21:
Is this running naked htrun, or via `yotta test`? If you're using yotta test, testing this branch might be useful, as it introduces additional logging, and a more careful approach to the way test output is processed: https://github.com/ARMmbed/yotta/pull/350

salkinium, 2015-09-21:
Yes, this is running via `yt test`, I'll try the branch.

salkinium, 2015-10-01:
The pyserial module on OSX seems to cause a reset to occur in the DAPLink when closing the connection. The DAPLink then buffers the output of the test, while the serial port is "disconnected".

After opening the connection again, flushing the input fails (for whatever reason), and the buffered output
is read into htrun, confusing it and sometimes failing prematurely.

The following simple code can be used to verify this behavior:
```python
import serial
ser = serial.Serial('/dev/tty.usbmodem1412', 9600)
ser.flushInput()
while True:
    print ser.readline()
```

We couldn't replicate this problem on Win7, so I assume this has something to do with how the serial drivers on OS X interact with DAPLink.

autopulated, 2015-10-01:
So trying more persistently to flush the output after opening the connection seems like the best workaround?

salkinium, 2015-10-01:
It's not `pyserial`s fault, picocom also resets the device on disconnection.

@autopulated I think you're right. The following crude workaround does flush the input.

```python
import serial
import time

ser = serial.Serial('/dev/tty.usbmodem1412', 9600)

ser.flushInput()
time.sleep(1)
ser.flushInput()
time.sleep(1)
ser.flushInput()

while True:
    print ser.readline()
```

You still need the `time.sleep(1)`, just calling `flushInput()` three times doesn't work.
The flushing only happens in the OS buffer, not the DAPLink (even though it should, but I don't know if that is possible).

salkinium, 2015-10-01:
Yay! This workaround also works in htrun!
However, only if you call `flushInput()` it doesn't work only with `flush()`.

PrzemekWirkus, 2015-10-02:
Have you tried:
```python
from subprocess import call
# ...
ser.flush()
call(['sync'], shell=True)
```

salkinium, 2015-10-02:
Yes, and it makes no difference :-(

autopulated, 2015-10-02:
Does it work with shorter sleep times, or even no sleep time, but just a sleep(0) (it's possible that sleep() is just yielding, and the delay isn't important)?

salkinium, 2015-10-09:
Interestingly, I do not get this behaviour when using the cmsis-dap chip from ST (called the st-linkv2-1 as found on the Nucleo boards).
I believe this is only on the combination of k64f and OS X.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSYST-188

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSYST-448

niklas-arm, 2015-12-11:
This seems to be fixed now with v0.1.18.

mazimkhan, 2016-01-20:
There are two issues mentioned here.
1. If the test has long serial output. It times out and serial output is cut off.
2. The unread serial output is buffered in cmsis-dap chip. This confuses mbedgt in next test/run and prematurely terminates the test.

Analysis on 1: If the test is long (i.e. how much time it takes to execute) then the timeout in the test should be set accordingly. In the gist I see ```HOST: Property 'timeout' = '20'```. This says terminate the test after 20 sec. Change it if you expect the test to run longer.

Analysis on 2: This is caused by greentea that terminates the test if it sees '{{end}}' in any part of the stdout from mbedhtrun. Even if it is printed in an error message. This is now fixed in https://github.com/ARMmbed/greentea/issues/70 PR https://github.com/ARMmbed/greentea/issues/71. This should fix this issue for now. However, a more robust implementation of target<->host communication may emerge in https://github.com/ARMmbed/greentea/issues/66.

Please share the feedback with version of greeantea.

closed 2016-03-11 by PrzemekWirkus