ARMmbed/htrun #26: [Win7] Test suite reports IOERR_SERIAL errors while testing K64F with firmware v0226

PrzemekWirkus, 2015-09-29:
# Description
* With old K64F firmware ```mbedhtrun``` behaves correctly not causing any serial port issues.
* With new firmware v0226 (bugfix for Linux users) for K64F platform ```mbedhtrun``` is not handling serial port operations correctly causing serial port access issues:
```
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -d E: -p COM149 -f ".\build\frdm-k64f-gcc\test\mbed-drivers-test-asynch_spi.bin" -C 4 -c shell -m K64F
mbedgt: mbed-host-test-runner: started
MBED: Instrumentation: "COM149" and disk: "E:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
........................................MBED: poll_for_serial_init(9600, 1, 40, 0.5, 0.25)
MBED: could not open port 'COM149': WindowsError(5, 'Access is denied.')

{{ioerr_serial}}
{{end}}
mbedgt: mbed-host-test-runner: stopped
mbedgt: mbed-host-test-runner: returned 'IOERR_SERIAL'
        test 'mbed-drivers-test-asynch_spi' .................................................... IOERR_SERIAL in 17.51 sec
```
See:
```
MBED: could not open port 'COM149': WindowsError(5, 'Access is denied.')
```

### K64F firmware (DETAILS.TXT dump):
```
Version: 0226
Build:   Aug 24 2015 17:06:30
Git Commit SHA: 27a236b9fe39c674a703c5c89655fbd26b8e27e1
Git Local mods: Yes
```

PrzemekWirkus, 2015-09-30:
Example for ```mbed-drivers```:
```
mbedgt: test report:
+---------------+---------------+------------------------------------+--------------+--------------------+-------------+
| target        | platform_name | test                               | result       | elapsed_time (sec) | copy_method |
+---------------+---------------+------------------------------------+--------------+--------------------+-------------+
| frdm-k64f-gcc | K64F          | mbed-drivers-test-echo             | OK           | 6.33               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-serial_interrupt | TIMEOUT      | 20.57              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-blinky           | IOERR_SERIAL | 17.21              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-functionpointer  | IOERR_SERIAL | 17.17              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-stdio            | IOERR_SERIAL | 17.48              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-eventhandler     | IOERR_SERIAL | 17.53              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-stl              | IOERR_SERIAL | 17.99              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-asynch_spi       | IOERR_SERIAL | 17.45              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-div              | IOERR_SERIAL | 17.17              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-rtc              | IOERR_SERIAL | 17.63              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-cstring          | IOERR_SERIAL | 17.48              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-cpp              | IOERR_SERIAL | 17.18              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-timeout          | IOERR_SERIAL | 17.13              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker_3         | IOERR_SERIAL | 17.13              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker_2         | IOERR_SERIAL | 17.12              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-heap_and_stack   | IOERR_SERIAL | 17.54              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-hello            | IOERR_SERIAL | 17.07              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-ticker           | IOERR_SERIAL | 17.17              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-dev_null         | IOERR_SERIAL | 17.71              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-basic            | IOERR_SERIAL | 17.17              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-time_us          | OK           | 11.38              | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-sleep_timeout    | IOERR_SERIAL | 17.2               | shell       |
| frdm-k64f-gcc | K64F          | mbed-drivers-test-detect           | IOERR_SERIAL | 17.13              | shell       |
+---------------+---------------+------------------------------------+--------------+--------------------+-------------+

Result: 2 OK / 20 IOERR_SERIAL / 1 TIMEOUT
Completed in 415.53 sec
```

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSYST-187

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSYST-447

PrzemekWirkus, 2015-11-04:
Fixed with [v0.1.17](https://github.com/ARMmbed/htrun/releases/tag/v.0.1.17)

closed 2015-11-04 by PrzemekWirkus