ARMmbed/htrun #28: mbed-host-test requires primary DNS suffix configured to computer name

mazimkhan, 2015-11-10:
OS: Windows 7
file htrun/mbed_host_tests/host_tests/test_socket_server_udp.py
line 24
Issue: socket.gethostbyname(socket.getfqdn()) fails to resolve to IP address

On Windows 7 call to socket.getfqdn() gives <computer>.<domain suffix>  (Fully qualified domain name)
In order for socket.gethostbyname() to work it expects that computer's full name is like <computer>.<domain suffix>.

The full computer name may not be in <computer>.<domain suffix> format. In that case socket.gethostbyname() fails to resolve socket.getfqdn() (<computer>.<domain suffix>) into the IP address.

Possible solutions:
By configuration: Follow the link for steps for configuring DNS suffix: https://technet.microsoft.com/en-us/library/cc786695(v=ws.10).aspx
Add this to documentation.

In code: If gethostbyname() fail for FQDN then try host name. As hostname resolution to IP succeeds.

rainierwolfcastle, 2015-11-13:
ARM Internal Ref: IOTSYST-576

PrzemekWirkus, 2015-11-18:
1. I will make sure that then imported and added to host test registry initialization like this is not executed unless test uses UDP sockets etc.
2. README.md has to be updated with this configuration information to support users.

PrzemekWirkus, 2015-11-19:
Fixed with https://github.com/ARMmbed/htrun/pull/31.

closed 2015-11-19 by PrzemekWirkus