ARMmbed/htrun #33: Switch -e (enumerate host tests) doesn't supoprt 'yotta test'

PrzemekWirkus, 2015-11-23:
Currently ```yotta test``` command executes in module subdirectory and ```mbedgt``` (greentea) can't deduce and enumerate host tests in default ```module/test/host_tests```.

How to reproduce:
```
$ cd C:\Work\mbed-drivers
$ yotta test
```
* ```yotat test``` execution directory of tests (mbedgt): ```c:\Work\mbed-drivers\build\frdm-k64f-gcc\test```
  * Default host test directory assumed by greentea: ```c:\Work\mbed-drivers\build\frdm-k64f-gcc\test\test\host_tests```
  * Should be: ```c:\Work\mbed-drivers\build\test\host_tests```

niklas-arm, 2015-12-11:
I can reproduce this issue with v0.1.18.

niklas-arm, 2015-12-14:
Note that this can be fixed by appending this to the `module.json`:

```json
  "scripts": {
    "testReporter": [
      "mbedgt",
      "--digest",
      "stdin",
      "-v",
      "-V",
      "-e", "test/host_tests"
    ]
  }
```

PrzemekWirkus, 2016-05-16:
Obsolete issue. Closing.

closed 2016-05-16 by PrzemekWirkus