ARMmbed/htrun #37: htrun - if a test timeouts, does it kill all running processes?

0xc0170, 2015-12-16:
I am running ``` mbedgt -V -n socket-test-test-tcp_client_echo_v0 ```. The test timeouts, and I can still see socket open (listening), and python process is running even though test time-outed.

Using mbedhtrun 0.1.18, mbedgt 0.1.14. I created an issue here, but might be mbedgt ? Let me know what do you think

0xc0170, 2015-12-17:
The  idea might be to have setup/teardown? If test method creates some resources, how do we expect cleanup at the moment?

PrzemekWirkus, 2015-12-22:
We have setup and teardown called ramUp and rampDown (to be compliant with other test library we call cli-test).

```mbedhtrun``` is the thing called in timeouted process or from command line so you control things there.

I can see you've already told us about setup/teardown naming convention you prefer.
I think it gones like this in host test:
1. call to rampUp() (setup)
2. call to test()
3. call to rampDown (teardown)

All calls are blocking so there is not timeout on those function execution inside ```mbedhtrun```. ```mbedhtrun``` is designe dto run in separate process executed by tools like greentea.

PrzemekWirkus, 2015-12-22:
@0xc0170 Doesn't create new processes. It only calls command line tools if plugin defines such behavior.
You just have to be sure that your implementation actually finish, you can't just create infinite loop in test method, you need to implement behavior responsible for tearing down your host test.

closed 2016-01-04 by PrzemekWirkus