ARMmbed/htrun #38: Copy all files within root directory - when using with -e option

0xc0170, 2015-12-16:
I am probably misusing htrun at the moment, but should not allow me to do this:

```
mbedhtrun -d E: -p COM64 -e test\host_tests\tcp_echo_client.py

HOST: Copy image onto target...
.\.gitignore
.\.yotta.json
.\module.json
.\README.md
        4 file(s) copied.
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Error! No host test name defined in preamble

{{error}}
{{end}}
HOST: Passive mode...
```

PrzemekWirkus, 2015-12-22:
Thanks for catching this! I will fix this, thanks! You are actually by "missuing it" finding lots of things not fixed, great work!

PS: Option ```-e``` should be used with directory, not file:
```
  -e ENUM_HOST_TESTS, --enum-host-tests=ENUM_HOST_TESTS
                        Define directory with local host tests
```

PrzemekWirkus, 2015-12-22:
Will be fixed in next release :P

closed 2016-01-04 by PrzemekWirkus