ARMmbed/htrun #39: mbed_client_auto test generates import errors on some platforms

bremoran, 2015-12-17:
This code is not guaranteed to work across platforms:
https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests/mbed_client_auto.py#L41

```Python
OWN_PC_ADDRESS = socket.gethostbyname(socket.gethostname())
```

Please use this instead:

```Python
SERVER_IP = str(socket.gethostbyname(socket.getfqdn()))`
```

PrzemekWirkus, 2015-12-22:
I think this test sets stuff during object attributes setting (similar to construction).
This is not allowed in this type of scrips. I will change it, but I think we should remove this host test from htrun soon.

PrzemekWirkus, 2015-12-22:
I've removed this host test from ```htrun``` because this type of tests should be mode to ```mymodule/test/host_tests```, vide: https://github.com/ARMmbed/htrun/pull/32

closed 2016-01-04 by PrzemekWirkus