ARMmbed/htrun #40: htrun fails to load on networks without DHCP/DNS integration

bremoran, 2015-12-17:
On networks without DHCP/DNS integration, htrun fails to load network tests, which causes all of htrun to fail to load. This is hidden by https://github.com/ARMmbed/greentea/issues/45.

The load failure is because hosts do not have resolvable names on networks without a DHCP/DNS integration, which causes the call to: ```gethostbyname``` to fail.

Many home networks do not have DHCP/DNS integration.

bremoran, 2015-12-17:
Many hosts have multiple network connections, which are not necessarily bridged. 
Many network tests require that the DUT is provided with the host's address via serial.

In these situations, it's possible that the current solution could provide the wrong result:

```Python
SERVER_IP = str(socket.gethostbyname(socket.getfqdn()))
```

This is particularly the case when there is a host that has two network interfaces:

* a private internal network with no internet access (a test network)
* an internet connected network

In this case, ```getfqdn``` will return the address of the first interface that has a fully qualified domain name.

A further limitation is for networks on which there is no DNS/DHCP integration, including many home networks. This causes ```getfqdn``` to return the computer name, which is not resolvable.

To resolve both of these problems, a network-based link discovery protocol should be used. This could take several forms, such as a UDP broadcast, however it seems prudent to use an existing network discovery protocol, like LLDP.

If LLDP were in use, the process would be:
1. mbed-gt flashes the new image
1. the DUT indicates that a network test is to be used
1. the host sends a unique ID to the DUT
1. the network test starts a LLDP listener
1. DUT starts a LLDP broadcast, indicating its IP address and the unique ID
1. The host captures the DUT's broadcast and sends the DUT the IP address of the interface that received the broadcast first (in some situations, multiple interfaces may receive the broadcast)

The process would look much the same with UDP broadcast, but LLDP would provide more long-term value, since it adds more capability to the mbed ecosystem.

PrzemekWirkus, 2016-01-04:
Brendan, thank you very much for this valuable comments. These valuable tips how to write proper networking host test are pretty important.
I've removed things that are using "bad way" mentioned here in this https://github.com/ARMmbed/htrun/pull/43.

In the future we will probably address this issue with external users.

closed 2016-01-04 by PrzemekWirkus