ARMmbed/htrun #41: What is a Rump Up?

niklas-arm, 2015-12-18:
https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests/base_host_test.py#L25

0xc0170, 2015-12-18:
Is this a proper naming for those 2 functions - rampUp, rampDown? 

We have setup and teardown in other tools.

niklas-arm, 2015-12-18:
:+1:  for setup and teardown, as they are more closely associated with testing.
RampUp and RampDown seems to be a business or engineering term, not usually associated with testing.

marhil01, 2015-12-18:
We have to check the naming in clitest if other our tools will have setup and teardown

niklas-arm, 2015-12-18:
There seems to be massive confusion in our python scripts:

- [Plenty of Ramps](https://github.com/search?utf8=✓&q=user%3Aarmmbed+language%3Apython+rampdown&type=Code).
- [Plenty of Teardowns](https://github.com/search?utf8=✓&q=user%3Aarmmbed+language%3Apython+teardown&type=Code).
- [I could not find any rampups or rampdowns in C or C++ code](https://github.com/search?utf8=✓&q=user%3Aarmmbed+language%3Ac+ramp&type=Code).
- [This is not surprising as "setup" and "teardown" are the **defacto standard terms** for testing](https://github.com/search?utf8=✓&q=user%3Aarmmbed+language%3Ac+teardown&type=Code).

cc @hugovincent @bogdanm

marhil01, 2015-12-21:
ok there will be added setUp and tearDown to clitest framework: IOTSYST-695

PR #42 (2015-12-22): rampUp, not rumpUp.
Fixes #41.
closed 2015-12-22 by PrzemekWirkus