ARMmbed/htrun #44: clear serial buffer before reading

LiyouZhou, 2015-12-30:
On a ubuntu platforms, it is found that remnants from previous runs of htrun in the serial buffer could messup the current run.

PrzemekWirkus, 2016-01-04:
Can you please give an example etc. I wish to reproduce and was wondering how I can do that?

niklas-arm, 2016-01-04:
Same on OS X when using K64F.
This makes `yt test` unusable on OSX.

This behavior doesn't happen with the ST-Linkv2-1, though, so this is a bug in the firmware of the interface chip on the K64F.
This `htrun` has to deal with this correctly.

LiyouZhou, 2016-01-04:
for `yt test` failure, I have filed an issue here: https://github.com/autopulated/mbed-test-wrapper/issues/2

LiyouZhou, 2016-01-04:
@PrzemekWirkus 

1. Run test normally, takes 16 seconds.
1. Run test and ctrl+c keyboard interrupt halfway at this point the board is still sending stuff to the host, leaving undigested stuff in the serial buffer.
1. Run test again, the test finds some stuff it not suppose to.

See detailed log below.

```
liyzho01@e105699-lin:/work/iot/minar$ mbedgt -t frdm-k64f-gcc -V -n minar-test-dispatchtest
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	detected 'K64F' -> 'K64F[0]', console at '/dev/ttyACM0', mounted at '/media/MBED', target id '024002261e501e62000000000000000000000000e389e3da'
mbedgt: local yotta target search in './yotta_targets' for compatible mbed-target 'k64f'
	inside './yotta_targets/frdm-k64f-gcc' found compatible target 'frdm-k64f-gcc'
mbedgt: processing 'frdm-k64f-gcc' yotta target compatible platforms...
mbedgt: processing 'K64F' platform...
mbedgt: using platform 'K64F' for test:
	target_id_mbed_htm = '024002261e501e62000000000000000000000000e389e3da'
	mount_point = '/media/MBED'
	target_id = '024002261e501e62000000000000000000000000e389e3da'
	serial_port = '/dev/ttyACM0'
	target_id_usb_id = '024002261e501e62000000000000000000000000e389e3da'
	platform_name = 'K64F'
	platform_name_unique = 'K64F[0]'
mbedgt: building your sources and tests with yotta...
	calling yotta: yotta --target=frdm-k64f-gcc,* build
info: generate for target: frdm-k64f-gcc 2.0.0 at /work/iot/minar/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /work/iot/minar/build/frdm-k64f-gcc
ninja: no work to do.
mbedgt: yotta build for target 'frdm-k64f-gcc' was successful
mbedgt: test case filter (specified with -n option)
	test filtered in 'minar-test-dispatchtest'
mbedgt: running 1 test for target 'frdm-k64f-gcc' and platform 'K64F'
	use 1 instance for testing
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -d /media/MBED -p /dev/ttyACM0:9600 -f "./build/frdm-k64f-gcc/test/minar-test-dispatchtest.bin" -C 4 -c shell -m K64F
mbedgt: mbed-host-test-runner: started
MBED: Instrumentation: "/dev/ttyACM0" and disk: "/media/MBED"
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
setting duration: 10
HOST: Detecting test case properties...
HOST: Property 'timeout' = '20'
setting duration auto timeout 20
HOST: Property 'host_test_name' = 'default'
HOST: Property 'description' = 'Dispatch test'
HOST: Property 'test_id' = 'DISPATCH_TEST'
HOST: Start test...
^Cmbedgt: ctrl+c keyboard interrupt!
Exception in thread Thread-2 (most likely raised during interpreter shutdown):liyzho01@e105699-lin:/work/iot/minar$ ^C
liyzho01@e105699-lin:/work/iot/minar$ ^C
liyzho01@e105699-lin:/work/iot/minar$ ^C
liyzho01@e105699-lin:/work/iot/minar$ mbedgt -t frdm-k64f-gcc -V -n minar-test-dispatchtest
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	detected 'K64F' -> 'K64F[0]', console at '/dev/ttyACM0', mounted at '/media/MBED', target id '024002261e501e62000000000000000000000000e389e3da'
mbedgt: local yotta target search in './yotta_targets' for compatible mbed-target 'k64f'
	inside './yotta_targets/frdm-k64f-gcc' found compatible target 'frdm-k64f-gcc'
mbedgt: processing 'frdm-k64f-gcc' yotta target compatible platforms...
mbedgt: processing 'K64F' platform...
mbedgt: using platform 'K64F' for test:
	target_id_mbed_htm = '024002261e501e62000000000000000000000000e389e3da'
	mount_point = '/media/MBED'
	target_id = '024002261e501e62000000000000000000000000e389e3da'
	serial_port = '/dev/ttyACM0'
	target_id_usb_id = '024002261e501e62000000000000000000000000e389e3da'
	platform_name = 'K64F'
	platform_name_unique = 'K64F[0]'
mbedgt: building your sources and tests with yotta...
	calling yotta: yotta --target=frdm-k64f-gcc,* build
info: generate for target: frdm-k64f-gcc 2.0.0 at /work/iot/minar/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /work/iot/minar/build/frdm-k64f-gcc
ninja: no work to do.
mbedgt: yotta build for target 'frdm-k64f-gcc' was successful
mbedgt: test case filter (specified with -n option)
	test filtered in 'minar-test-dispatchtest'
mbedgt: running 1 test for target 'frdm-k64f-gcc' and platform 'K64F'
	use 1 instance for testing
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -d /media/MBED -p /dev/ttyACM0:9600 -f "./build/frdm-k64f-gcc/test/minar-test-dispatchtest.bin" -C 4 -c shell -m K64F
mbedgt: mbed-host-test-runner: started
MBED: Instrumentation: "/dev/ttyACM0" and disk: "/media/MBED"
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
setting duration: 10
HOST: Detecting test case properties...
HOST: Unknown test property print format: {success}}
HOST: Unknown test property print format: {{end}}
Hmbedgt: mbed-host-test-runner: stopped
mbedgt: mbed-host-test-runner: returned 'OK'
mbedgt: test on hardware with target id: 024002261e501e62000000000000000000000000e389e3da 
	test 'minar-test-dispatchtest' ......................................................... OK in 0.12 sec
mbedgt: test report:
+---------------+---------------+-------------------------+--------+--------------------+-------------+
| target        | platform_name | test                    | result | elapsed_time (sec) | copy_method |
+---------------+---------------+-------------------------+--------+--------------------+-------------+
| frdm-k64f-gcc | K64F          | minar-test-dispatchtest | OK     | 0.12               | shell       |
+---------------+---------------+-------------------------+--------+--------------------+-------------+

Result: 1 OK
completed in 9.63 sec
liyzho01@e105699-lin:/work/iot/minar$ 
```

PrzemekWirkus, 2016-01-04:
Thanks guys!

mazimkhan, 2016-01-20:
This seems to be an issue in greentea that terminates a test even if ```{{end}}``` appears in an error message. 
Solution is that greentea should only end a test when ```^{{end}}$``` is received. This solution is within the premise of current implementation. In future we may need a simple test case state machine that is driven by communication between target and host test sides on the serial port.

mazimkhan, 2016-01-20:
@LiyouZhou please test https://github.com/ARMmbed/greentea/pull/71/

mazimkhan, 2016-01-20:
Closing this as issue created in greentea https://github.com/ARMmbed/greentea/issues/70

closed 2016-01-20 by mazimkhan