ARMmbed/htrun #49: Test runner - test() async support

0xc0170, 2016-01-07:
The code reference for what I understand is the test runner which expects the result to be returned from a test(): https://github.com/ARMmbed/htrun/blob/a8435d25e66f6d5232460e55f27e978fced1887b/mbed_host_tests/__init__.py#L410

Shall we consider async support for our test cases (for host tests)? This might be useful to write async host test, which would be terminated once {{end}} is received. 

For instance, client has 2 test cases, which send data. Host test echoes them back. Once client is done and the tests passes, sends {{end}}, this would terminate host test() method, call teardown().

0xc0170, 2016-01-08:
The code which we wrote was referenced here : https://github.com/ARMmbed/greentea/issues/50 (it's not quite correct but illustrates a user case for having async host test).

PrzemekWirkus, 2016-02-09:
This will be covered by this breaking change: https://github.com/ARMmbed/greentea/pull/78

PrzemekWirkus, 2016-02-09:
I will close this because we will continue in separate threads.

closed 2016-02-09 by PrzemekWirkus