ARMmbed/htrun #60: Missing setup() / teardown() call in main htrun flow

PrzemekWirkus, 2016-01-20:
# Description
We are [only calling ```setup()``` / ```teardown()```](https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/__init__.py#L413) when we handle ```---run``` switch (not used in CI). We are not calling setup/teardown in normal workflow:
```python
def execute_run(self):
...
try:
    self.test_supervisor = get_host_test("run_binary_auto")
    # Call to setup if function is implemented
    if hasattr(self.test_supervisor, 'setup') and callable(getattr(self.test_supervisor, 'setup')):
        self.test_supervisor.setup()

    # Call to test function
    result = self.test_supervisor.test(self)    # This is blocking, waits for {end}

    # Call to teardown if function is implemented
    if hasattr(self.test_supervisor, 'teardown') and callable(getattr(self.test_supervisor, 'teardown')):
        self.test_supervisor.teardown()
except Exception, e:
    print str(e)
    self.print_result(self.RESULT_ERROR)
```

We should be calling it also [here](https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/__init__.py#L480):
```python
def execute(self):
...
try:
    CONFIG = self.detect_test_config(verbose=True) # print CONFIG

    result = None
    if "host_test_name" in CONFIG:
        if is_host_test(CONFIG["host_test_name"]):
            #self.notify("HOST: CONFIG['host_test_name'] is '%s'" % CONFIG["host_test_name"])
            self.test_supervisor = get_host_test(CONFIG["host_test_name"])
            result = self.test_supervisor.test(self)    #result = self.test()
        else:
            self.notify("HOST: Error! Unknown host test name '%s' (use 'mbedhtrun --list' to verify)!"% CONFIG["host_test_name"])
            self.notify("HOST: Error! You can use switch '-e <dir>' to specify local directory with host tests to load")
            self.print_result(self.RESULT_ERROR)
    else:
        self.notify("HOST: Error! No host test name defined in preamble")
        self.print_result(self.RESULT_ERROR)
```

probably missing code in above snippet:
```python
self.test_supervisor.setup()
result = self.test_supervisor.test(self)    #result = self.test()
self.test_supervisor.teardown()
```

mazimkhan, 2016-01-20:
Calling ```self.test_supervisor.teardown()``` in a try: finally would be a good idea.

PrzemekWirkus, 2016-01-20:
@mazimkhan Do you think that there is a chance for us to call teardown twice?
* Once we try to kill a process and we already called a teardown in a setup-test-teardown flow?

mazimkhan, 2016-01-20:
@PrzemekWirkus yes it is possible that's why in abort () any exceptions raising from it are suppressed.

mazimkhan, 2016-01-20:
@PrzemekWirkus moreover when greentea decides to terminate mbedhtrun it has already failed the test and stopped reading mbedhtrun stdout. Main requirement in that scenario is to indicate to the test to cleanup/unblock.

PR #63 (2016-01-21): Add missing setup/teardown call
fix #60
closed 2016-01-21 by PrzemekWirkus