ARMmbed/htrun #79: mbed-drivers-test-echo test fails

ccli8, 2016-04-01:
Hi,

mbed-drivers-test-echo contains one test suite which contains three test cases.
In the first test case, mbed-host-tests will send one more KV pair than specified and interfere with next test case. So the second test case fails. This issue is met on Nuvoton's board but not on K64F board. I check with K64F and find mbed-host-tests still send one more KV pair as above, but I don't know why K64F can pass it.

Test environment
```
mbed-drivers: 1.4.0
mbed-host-tests: 0.2.6
```
Test flow
```
mbedgt -n mbed-drivers-test-echo -V
```
Test log
```
[1459490609.50][HTST][INF] starting host test process...
[1459490609.95][CONN][INF] starting connection process...
[1459490609.95][CONN][INF] initializing serial port listener...
[1459490609.95][SERI][INF] serial(port=COM31, baudrate=9600)
[1459490609.95][SERI][INF] reset device using 'default' plugin...
[1459490610.22][SERI][INF] wait for it...
[1459490611.23][CONN][INF] sending preamble 'a6d4ccb1-30c7-4a68-ba76-ebc8e51525d2'...
[1459490611.23][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1459490611.23][SERI][TXD] {{__sync;a6d4ccb1-30c7-4a68-ba76-ebc8e51525d2}}
[1459490611.37][CONN][INF] found SYNC in stream: {{__sync;a6d4ccb1-30c7-4a68-ba76-ebc8e51525d2}}, queued...
[1459490611.37][HTST][INF] sync KV found, uuid=a6d4ccb1-30c7-4a68-ba76-ebc8e51525d2, timestamp=1459490611.373000
[1459490611.37][CONN][RXD] {{__sync;a6d4ccb1-30c7-4a68-ba76-ebc8e51525d2}}
[1459490611.39][CONN][INF] found KV pair in stream: {{__version;1.0.0}}, queued...
[1459490611.39][HTST][INF] DUT greentea-client version: 1.0.0
[1459490611.39][CONN][RXD] {{__version;1.0.0}}
[1459490611.40][CONN][INF] found KV pair in stream: {{__timeout;10}}, queued...
[1459490611.40][HTST][INF] setting timeout to: 10 sec
[1459490611.40][CONN][RXD] {{__timeout;10}}
[1459490611.43][CONN][INF] found KV pair in stream: {{__host_test_name;echo}}, queued...
[1459490611.43][HTST][INF] host test setup() call...
[1459490611.43][HTST][INF] CALLBACKs updated
[1459490611.43][HTST][INF] host test detected: echo
[1459490611.43][CONN][RXD] {{__host_test_name;echo}}
[1459490611.45][CONN][INF] found KV pair in stream: {{__testcase_count;3}}, queued...
[1459490611.45][CONN][RXD] {{__testcase_count;3}}
[1459490611.48][CONN][RXD] >>> Running 3 test cases...
[1459490611.50][CONN][INF] found KV pair in stream: {{echo_count;16}}, queued...
[1459490611.50][SERI][TXD] {{echo_count;16}}
[1459490611.50][CONN][RXD] {{echo_count;16}}
[1459490611.50][SERI][TXD] {{echo;918a603b-f5f4-4c21-ad6d-c280cbb44622}}
[1459490611.62][CONN][INF] found KV pair in stream: {{echo;918a603b-f5f4-4c21-ad6d-c280cbb44622}}, queued...
[1459490611.62][SERI][TXD] {{echo;f8a44d38-3513-4a16-971b-2509404387d3}}
[1459490611.62][CONN][RXD] {{echo;918a603b-f5f4-4c21-ad6d-c280cbb44622}}
[1459490611.72][CONN][INF] found KV pair in stream: {{echo;f8a44d38-3513-4a16-971b-2509404387d3}}, queued...
[1459490611.72][SERI][TXD] {{echo;ef663022-f2a4-455e-9b6f-989892399b52}}
[1459490611.72][CONN][RXD] {{echo;f8a44d38-3513-4a16-971b-2509404387d3}}
[1459490611.81][CONN][INF] found KV pair in stream: {{echo;ef663022-f2a4-455e-9b6f-989892399b52}}, queued...
[1459490611.81][CONN][RXD] {{echo;ef663022-f2a4-455e-9b6f-989892399b52}}
[1459490611.81][SERI][TXD] {{echo;77b30340-e301-44b6-9649-f50e3f3806d5}}
[1459490611.90][CONN][INF] found KV pair in stream: {{echo;77b30340-e301-44b6-9649-f50e3f3806d5}}, queued...
[1459490611.90][SERI][TXD] {{echo;eb36ac12-b56d-445a-8c67-cc70c06607aa}}
[1459490611.90][CONN][RXD] {{echo;77b30340-e301-44b6-9649-f50e3f3806d5}}
[1459490612.00][CONN][INF] found KV pair in stream: {{echo;eb36ac12-b56d-445a-8c67-cc70c06607aa}}, queued...
[1459490612.00][SERI][TXD] {{echo;1e024901-98ff-4622-b335-603cbffba496}}
[1459490612.00][CONN][RXD] {{echo;eb36ac12-b56d-445a-8c67-cc70c06607aa}}
[1459490612.09][CONN][INF] found KV pair in stream: {{echo;1e024901-98ff-4622-b335-603cbffba496}}, queued...
[1459490612.09][SERI][TXD] {{echo;7e5e1742-0c89-4218-b606-67e41ada8406}}
[1459490612.09][CONN][RXD] {{echo;1e024901-98ff-4622-b335-603cbffba496}}
[1459490612.18][CONN][INF] found KV pair in stream: {{echo;7e5e1742-0c89-4218-b606-67e41ada8406}}, queued...
[1459490612.18][SERI][TXD] {{echo;050ad3e1-e610-4fc1-8024-4b40b10c1655}}
[1459490612.18][CONN][RXD] {{echo;7e5e1742-0c89-4218-b606-67e41ada8406}}
[1459490612.28][CONN][INF] found KV pair in stream: {{echo;050ad3e1-e610-4fc1-8024-4b40b10c1655}}, queued...
[1459490612.28][SERI][TXD] {{echo;aa065b99-f197-4577-a43b-2efe4dd51a3f}}
[1459490612.28][CONN][RXD] {{echo;050ad3e1-e610-4fc1-8024-4b40b10c1655}}
[1459490612.37][CONN][INF] found KV pair in stream: {{echo;aa065b99-f197-4577-a43b-2efe4dd51a3f}}, queued...
[1459490612.37][SERI][TXD] {{echo;a324735f-621c-4b11-86b5-e79f38e6eaaa}}
[1459490612.37][CONN][RXD] {{echo;aa065b99-f197-4577-a43b-2efe4dd51a3f}}
[1459490612.46][CONN][INF] found KV pair in stream: {{echo;a324735f-621c-4b11-86b5-e79f38e6eaaa}}, queued...
[1459490612.46][SERI][TXD] {{echo;3606d3d6-e2d3-4a49-95f2-cbc1e92f7c52}}
[1459490612.46][CONN][RXD] {{echo;a324735f-621c-4b11-86b5-e79f38e6eaaa}}
[1459490612.56][CONN][INF] found KV pair in stream: {{echo;3606d3d6-e2d3-4a49-95f2-cbc1e92f7c52}}, queued...
[1459490612.56][SERI][TXD] {{echo;b3f9b48e-8983-43f6-95a8-132dd2329013}}
[1459490612.56][CONN][RXD] {{echo;3606d3d6-e2d3-4a49-95f2-cbc1e92f7c52}}
[1459490612.65][CONN][INF] found KV pair in stream: {{echo;b3f9b48e-8983-43f6-95a8-132dd2329013}}, queued...
[1459490612.67][SERI][TXD] {{echo;8af73d75-6609-4513-ba3e-4fe4b829bb53}}
[1459490612.67][CONN][RXD] {{echo;b3f9b48e-8983-43f6-95a8-132dd2329013}}
[1459490612.76][CONN][INF] found KV pair in stream: {{echo;8af73d75-6609-4513-ba3e-4fe4b829bb53}}, queued...
[1459490612.76][SERI][TXD] {{echo;cbcce0be-c7b8-4222-8096-62b7a820cf33}}
[1459490612.76][CONN][RXD] {{echo;8af73d75-6609-4513-ba3e-4fe4b829bb53}}
[1459490612.86][CONN][INF] found KV pair in stream: {{echo;cbcce0be-c7b8-4222-8096-62b7a820cf33}}, queued...
[1459490612.86][SERI][TXD] {{echo;b2b13b58-ed3a-4c0a-af0b-3fed4ab6c9f5}}
[1459490612.86][CONN][RXD] {{echo;cbcce0be-c7b8-4222-8096-62b7a820cf33}}
[1459490612.95][CONN][INF] found KV pair in stream: {{echo;b2b13b58-ed3a-4c0a-af0b-3fed4ab6c9f5}}, queued...
[1459490612.95][SERI][TXD] {{echo;d0476bc5-2622-45a7-b7da-71d335aa25f9}}
[1459490612.95][CONN][RXD] {{echo;b2b13b58-ed3a-4c0a-af0b-3fed4ab6c9f5}}
[1459490613.04][CONN][INF] found KV pair in stream: {{echo;d0476bc5-2622-45a7-b7da-71d335aa25f9}}, queued...
[1459490613.04][SERI][TXD] {{echo;0ea5f1f4-168f-48c3-b7b2-dee97cbd1f96}}
[1459490613.04][CONN][RXD] {{echo;d0476bc5-2622-45a7-b7da-71d335aa25f9}}
[1459490613.06][CONN][INF] found KV pair in stream: {{echo_count;32}}, queued...
[1459490613.06][SERI][TXD] {{echo_count;32}}
[1459490613.06][SERI][TXD] {{echo;7317ef92-fe8e-4d88-8363-c678031f0c61}}
[1459490613.06][CONN][RXD] {{echo_count;32}}
[1459490613.12][CONN][RXD] :36::FAIL: Expected 32 Was 0
[1459490613.18][CONN][RXD] >>> failure with reason 'Assertion Failed' during 'Case Handler'
[1459490621.42][HTST][INF] test suite run finished after 10.01 sec...
[1459490621.42][HTST][INF] CONN exited with code: 0
[1459490621.42][HTST][INF] No events in queue
[1459490621.42][HTST][INF] stopped consuming events
[1459490621.42][HTST][INF] host test result(): False
[1459490621.42][HTST][WRN] missing __exit event from DUT
[1459490621.42][HTST][INF] calling blocking teardown()
[1459490621.42][HTST][INF] teardown() finished
[1459490621.42][HTST][INF] {{result;failure}}
```

PrzemekWirkus, 2016-04-04:
@ccli8 Which compiler (yotta target) you've used to build mbed-driver ?

ccli8, 2016-04-06:
Hi,
Compiler: armcc 5.06 (MDK-ARM 5.17)
Board: Nuvoton's Nu-mbed-NUC472, but it is under test and not yet released
OS: Windows 7

```
mbedgt --version
0.2.8
mbedhtrun --version
0.2.6
mbedls --version
0.2.4
```

PrzemekWirkus, 2016-04-07:
@ccli8 OK, sounds legit ;) I will have a look soon at this issue.

PrzemekWirkus, 2016-04-08:
@ccli8 Can you check if [this simple fix](https://github.com/ARMmbed/htrun/pull/82) will help in your case?

ccli8, 2016-04-11:
@PrzemekWirkus Yes, it fix the issue. Thanks.

closed 2016-04-11 by ccli8

PrzemekWirkus, 2016-04-18:
@ccli8 Thank you ! :)