ARMmbed/htrun #87: Command options and the default behaviors

bridadan, 2016-05-08:
Pull requests #85 and #86 add support for checking the mount point and serial port for a platform. Since this information can be derived from mbed-ls (when supplied a target ID), it seems like the only time you would provide the serial port or mount point is when you don't know the target ID, the platform doesn't have a valid target ID, or you want to override one or many of the values that were resolved through mbed-ls.

If no target ID is provided, then it should default to requiring a provided serial port, baud rate, and mount point. If a target ID is provided, then you should be able to override the serial port, baud rate, and mount point through command line arguments.

It also seems valid to say sometimes you would want to override the baud rate without providing a serial port as part of the `-p` option (especially if you don't know the serial port). I would suggest adding a `--baud-rate` options to specify the serial port's baud rate specifically. The `-p` can still accept a baud rate component (`<port>:<baud rate>`) for backward compatibility. In the case that `-p <port>:<baud rate> --baud-rate <different baud rate>` are both specified, then `--baud-rate` would be used.

The priority for platform options from **lowest to highest**:

1. Values determined from target ID and mbed ls (with baud rate defaulting to 115200 or 9600)
2. `-d,--disk-path` for disk/mount point, `-p <port>:<baud rate>` for serial port (`:<baud rate>` bit being optional)
3. `--baud-rate` for serial baud rate

PrzemekWirkus, 2016-05-09:
Yes, this makes perfcet sense, we can iterate and change how -t behaves.

+1

mazimkhan, 2016-09-16:
@bridadan this issue is fixed here. We would need changes in Greentea to make use of this feature.

theotherjimmy, 2017-09-19:
@bridadan please reopen if this is still a problem.

closed 2017-09-19 by theotherjimmy