ARMmbed/htrun #120: Greentea passing test cases even when there is sync failure.

mazimkhan, 2016-08-04:
Below log shows that host test has failed to receive the sync correctly but since greentea works on test start and end prints it reads tests as pass. Perhaps it should make sure sync has worked correctly before parsing test case logs.

```
12:40:33 [1470311095.82][HTST][INF] starting host test process...
12:40:34 [1470311096.42][CONN][INF] starting serial connection process...
12:40:34 [1470311096.42][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
12:40:34 [1470311096.42][CONN][INF] initializing serial port listener...
12:40:34 [1470311096.42][HTST][INF] setting timeout to: 60 sec
12:40:34 [1470311096.42][SERI][INF] serial(port=COM18, baudrate=9600, timeout=0)
12:40:34 Plugin info: HostTestPluginBase::BasePlugin: Waiting up to 60 sec for '0240000029304e45002a500878a200311951000097969900' serial port (current is 'COM18')...
12:40:34 [1470311096.52][SERI][INF] reset device using 'default' plugin...
12:40:34 [1470311096.78][SERI][INF] waiting 1.00 sec after reset
12:40:35 [1470311097.78][SERI][INF] wait for it...
12:40:35 [1470311097.78][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
12:40:35 [1470311097.78][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
12:40:35 [1470311097.78][CONN][INF] sending preamble '4cb117f6-44e1-481d-b583-1093bab337cf'
12:40:35 [1470311097.78][SERI][TXD] {{__sync;4cb117f6-44e1-481d-b583-1093bab337cf}}
12:40:35 [1470311097.78][CONN][RXD] estcase_count;1}}
12:40:35 [1470311097.78][CONN][RXD] >>> Running 1 test cases...
12:40:35 [1470311097.92][CONN][RXD] {{__testcase_name;{{__sync;4cb117f6-44e1-481d-b583-1093bab337cf}}
12:40:35 [1470311097.92][CONN][WRN] found KV pair in stream: {{__testcase_name;{{__sync;4cb117f6-44e1-481d-b583-1093bab337cf}}, ignoring...
12:40:35 [1470311097.94][CONN][RXD] {{__version;1.3.0}}
12:40:35 [1470311097.94][CONN][WRN] found KV pair in stream: {{__version;1.3.0}}, ignoring...
12:40:35 [1470311097.95][CONN][RXD] {{__timeout;5}}
12:40:35 [1470311097.96][CONN][WRN] found KV pair in stream: {{__timeout;5}}, ignoring...
12:40:35 [1470311097.99][CONN][RXD] {{__host_test_name;default_auto}}
12:40:35 [1470311097.99][CONN][WRN] found KV pair in stream: {{__host_test_name;default_auto}}, ignoring...
12:40:35 [1470311098.02][CONN][RXD] {{__testcase_count;8}}
12:40:35 [1470311098.02][CONN][WRN] found KV pair in stream: {{__testcase_count;8}}, ignoring...
12:40:35 [1470311098.05][CONN][RXD] >>> Running 8 test cases...
12:40:35 [1470311098.08][CONN][RXD] {{__testcase_name;C strings: strtok}}
12:40:35 [1470311098.08][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: strtok}}, ignoring...
12:40:35 [1470311098.12][CONN][RXD] {{__testcase_name;C strings: strpbrk}}
12:40:35 [1470311098.12][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: strpbrk}}, ignoring...
12:40:35 [1470311098.18][CONN][RXD] {{__testcase_name;C strings: %i %d integer formatting}}
12:40:35 [1470311098.18][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: %i %d integer formatting}}, ignoring...
12:40:35 [1470311098.24][CONN][RXD] {{__testcase_name;C strings: %u %d integer formatting}}
12:40:35 [1470311098.24][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: %u %d integer formatting}}, ignoring...
12:40:35 [1470311098.30][CONN][RXD] {{__testcase_name;C strings: %x %E integer formatting}}
12:40:35 [1470311098.30][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: %x %E integer formatting}}, ignoring...
12:40:36 [1470311098.36][CONN][RXD] {{__testcase_name;C strings: %f %f float formatting}}
12:40:36 [1470311098.36][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: %f %f float formatting}}, ignoring...
12:40:36 [1470311098.41][CONN][RXD] {{__testcase_name;C strings: %e %E float formatting}}
12:40:36 [1470311098.41][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: %e %E float formatting}}, ignoring...
12:40:36 [1470311098.47][CONN][RXD] {{__testcase_name;C strings: %g %g float formatting}}
12:40:36 [1470311098.47][CONN][WRN] found KV pair in stream: {{__testcase_name;C strings: %g %g float formatting}}, ignoring...
12:40:36 [1470311098.52][CONN][RXD] >>> Running case #1: 'C strings: strtok'...
12:40:36 [1470311098.56][CONN][RXD] {{__testcase_start;C strings: strtok}}
12:40:36 [1470311098.56][CONN][WRN] found KV pair in stream: {{__testcase_start;C strings: strtok}}, ignoring...
12:40:36 [1470311098.60][CONN][RXD] {{__testcase_finish;C strings: strtok;1;0}}
12:40:36 [1470311098.60][CONN][WRN] found KV pair in stream: {{__testcase_finish;C strings: strtok;1;0}}, ignoring...
12:40:36 [1470311098.65][CONN][RXD] >>> 'C strings: strtok': 1 passed, 0 failed
12:40:36 [1470311098.70][CONN][RXD] >>> Running case #2: 'C strings: strpbrk'...
12:40:36 [1470311098.74][CONN][RXD] {{__testcase_start;C strings: strpbrk}}
12:40:36 [1470311098.74][CONN][WRN] found KV pair in stream: {{__testcase_start;C strings: strpbrk}}, ignoring...
12:40:36 [1470311098.78][CONN][INF] resending new preamble 'b028c99b-5297-434f-be73-1de8d47abbb4' after 1.00 sec
12:40:36 [1470311098.78][SERI][TXD] {{__sync;b028c99b-5297-434f-be73-1de8d47abbb4}}
12:40:36 [1470311098.78][CONN][RXD] {{__testcase_finish;C strings: strpbrk;1;0}}
12:40:36 [1470311098.78][CONN][WRN] found KV pair in stream: {{__testcase_finish;C strings: strpbrk;1;0}}, ignoring...
12:40:36 [1470311098.83][CONN][RXD] >>> 'C strings: strpbrk': 1 passed, 0 failed
12:40:36 [1470311098.90][CONN][RXD] >>> Running case #3: 'C strings: %i %d integer formatting'...
12:40:36 [1470311098.96][CONN][RXD] {{__testcase_start;C strings: %i %d integer formatting}}
12:40:36 [1470311098.96][CONN][WRN] found KV pair in stream: {{__testcase_start;C strings: %i %d integer formatting}}, ignoring...
12:40:36 [1470311099.02][CONN][RXD] {{__testcase_finish;C strings: %i %d integer formatting;1;0}}
12:40:36 [1470311099.02][CONN][WRN] found KV pair in stream: {{__testcase_finish;C strings: %i %d integer formatting;1;0}}, ignoring...
12:40:36 [1470311099.08][CONN][RXD] >>> 'C strings: %i %d integer formatting': 1 passed, 0 failed
12:40:36 [1470311099.15][CONN][RXD] >>> Running case #4: 'C strings: %u %d integer formatting'...
12:40:36 [1470311099.21][CONN][RXD] {{__testcase_start;C strings: %u %d integer formatting}}
12:40:36 [1470311099.21][CONN][WRN] found KV pair in stream: {{__testcase_start;C strings: %u %d integer formatting}}, ignoring...
12:40:36 [1470311099.28][CONN][RXD] {{__testcase_finish;C strings: %u %d integer formatting;1;0}}
[... 33 lines cut ...]
12:41:34 [1470311156.42][HTST][INF] test suite run finished after 60.00 sec...
12:41:34 [1470311156.42][CONN][INF] received special even '__host_test_finished' value='True', finishing
12:41:34 [1470311156.43][HTST][INF] CONN exited with code: 0
12:41:34 [1470311156.43][HTST][INF] No events in queue
12:41:34 [1470311156.43][HTST][INF] stopped consuming events
12:41:34 [1470311156.43][HTST][INF] host test result(): None
12:41:34 [1470311156.43][HTST][WRN] missing __exit event from DUT
12:41:34 [1470311156.43][HTST][ERR] missing __exit event from DUT and no result from host test, timeout...
12:41:34 [1470311156.43][HTST][INF] calling blocking teardown()
12:41:34 [1470311156.43][HTST][INF] teardown() finished
12:41:34 [1470311156.43][HTST][INF] {{result;timeout}}
12:41:34 mbedgt: checking for GCOV data...
12:41:34 mbedgt: mbed-host-test-runner: stopped
12:41:34 mbedgt: mbed-host-test-runner: returned 'TIMEOUT'
12:41:34 mbedgt: test on hardware with target id: 0240000029304e45002a500878a200311951000097969900
12:41:34 mbedgt: test suite 'tests-mbed_drivers-c_strings' .................................................... TIMEOUT in 70.51 sec
12:41:34 	test case: 'C strings: %e %E float formatting' ............................................... OK in 0.06 sec
12:41:34 	test case: 'C strings: %f %f float formatting' ............................................... OK in 0.07 sec
12:41:34 	test case: 'C strings: %g %g float formatting' ............................................... OK in 0.06 sec
12:41:34 	test case: 'C strings: %i %d integer formatting' ............................................. OK in 0.06 sec
12:41:34 	test case: 'C strings: %u %d integer formatting' ............................................. OK in 0.07 sec
12:41:34 	test case: 'C strings: %x %E integer formatting' ............................................. OK in 0.07 sec
12:41:34 	test case: 'C strings: strpbrk' .............................................................. OK in 0.04 sec
12:41:34 	test case: 'C strings: strtok' ............................................................... OK in 0.04 sec
```

theotherjimmy, 2017-09-19:
> but since greentea works ...

Then this is an issue for greentea right?

mazimkhan, 2017-09-19:
htrun sits between target and Greentea. Greentea can't detect this issue. However, this is pretty old. Hence need to be reproduced and impact has to be evaluated.

theotherjimmy, 2017-09-19:
Agreed. I'll take a look to see If I can reproduce it.

bridadan, 2019-01-07:
@theotherjimmy This is quite old, do remember by chance if you ever reproduced this?

bridadan, 2019-01-07:
Actually, I'm pretty sure we fixed this a long time ago. If a sync isn't successful it will retry and then fail after a timeout.

closed 2019-01-07 by bridadan