ARMmbed/htrun #144: Be more robust in log parsing

Patater, 2017-01-24:
Consider the following example logs.

### Good log

```
** Hello Example **
Main loop count: 0
Main loop count: 1
Main loop count: 2
Main loop count: 3
Main loop count: 4
Main loop count: 5
Main loop count: 6
Main loop count: 7
Main loop count: 8
Main loop count: 9
Main loop count: 10
IRQ 1 from Button 1 Pressed
Main loop count: 11
Main loop count: 12
Main loop count: 13
Main loop count: 14
Main loop count: 15
```

### Bad log
```
** Hello Example **
Main loop count: 0
Main loop count: 1
Main loop count: 2
Main loop count: 3
Main loop count: 4
Main loop count: 5
** Out of memory
Main loop count: 6
Main loop count: 7
Main loop count: 8
Main loop count: 9
Main loop count: 10
IRQ 1 from Button 1 Pressed
Main loop count: 11
Main loop count: 12
Main loop count: 13
Main loop count: 14
Main loop count: 15
```

### Good log
```
** Hello Example **
Main loop count: 0
Main loop count: 1
Main loop count: 2
Main loop count: 3
Main loop count: 4
Main loop count: 5
Main loop count: 6
Main loop count: 7
Main loop count: 8
Main loop count: 9
Main loop count: 10
Main loop count: 11
IRQ 1 from Button 1 Pressed
Main loop count: 12
Main loop count: 13
Main loop count: 14
Main loop count: 15
```

These logs are impossible to match with the current log matching functionality. The current log matching feature (for both programmer implementation and user cognitive overhead) would be simplified by matching with a single multi-line regular expression.

### Example log matcher
The following multi-line regular expression would deal with the asynchronicity of the previous logs, where the IRQ could come at basically any time relative to the "Main loop count" lines. It would also fail to match if the `** Out of memory` print happened at any time from some other thread in the system.

(\*\* Hello Example \*\*\r\n)(Main loop count: \d+\r\n)+(IRQ \d+ from Button \d+ Pressed\r\n)(Main loop count: \d+\r\n)

mazimkhan, 2017-01-25:
This comment compares two approaches for working out a better approach. Please see details of the existing feature [here](https://github.com/ARMmbed/htrun#testing-mbed-os-examples)

| Scenarios\Approaches | Existing  | Multi-line Regex  |
|---|---|---|
| Simple Deterministic log: Where order and contents of log lines do not change  | Literally matching lines works fine.   | You have to write in regex syntax. As big as the no. of lines. |
| Special characters  | Since existing feature first tries to literally match each line we don't need to escape the special characters.  | This will need escaping special characters. Effort proportional to no. of lines. |
| Lines in specific order  | Just put the lines (literal/regex) in desired order and they would be matched in that order. | Supported by multi line regex  |
| *Filtering unwanted lines/text | Not supported and known short coming. It expects examples to have some text that would only appear if no errors were encountered. | Multi-line regex will fail if a non expecting line appears in between. |

\*This issue relates to the last scenario above.

Patater, 2017-01-25:
It also applies to issues where examples have non-deterministic line orderings (like in the IRQ example provided above).

theotherjimmy, 2017-09-19:
@Patater @mazimkhan How's this issue going? should we consider an enhancement to htrun?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-82*

theotherjimmy, 2018-10-04:
@Patater Thanks for responding. I'll make sure this ends up in my backlog. Does this affect your ability to test certain examples? Do you have a easy-to-use reproducer that you would like to be able to test?

Patater, 2018-10-04:
This was originally for a uVisor example, but I'm not working on that project anymore.

If you want to reproduce easily, I think just a printf of what I put in the description would work. That lets you test the regex. If you really want to reproduce something slightly more realistic, extend blinky to print on every blink, and to also print when a user presses a button that triggers and IRQ that sets a semaphore that wakes a thread. Then test and press the button at different points in time.

CC @alzix

theotherjimmy, 2018-10-04:
Thanks. If it's not blocking you right now, I'm going to put it off to work on things that are blocking other people. Sorry about that.