ARMmbed/htrun #156: Connections to RaaS often timeout

marcuschangarm, 2017-05-12:
When invoking htrun through greentea, the connection to RaaS often timeout. This makes it difficult to maintain a proper CI flow.

Greentea called from Jenkins:
```
execute("mbedgt --grm ${target}:raas_client:54.229.57.51:8000 -V -v --test-spec ${build_dir}/test_spec.json -n ${test_filters.join(',')} -e TESTS/powercut/host_tests")
```

Output from failed run:
```
10:42:08 [K64F GCC_ARM default] [1494585728.90][requests.packages.urllib3.connectionpool]Starting new HTTP connection (1): 54.229.57.51
10:42:08 [K64F GCC_ARM default] [1494585728.97][requests.packages.urllib3.connectionpool]http://54.229.57.51:8000 "PUT /resource/0240000034544e45002d00028aa900022011000097969900/disconnect HTTP/1.1" 200 85
10:42:08 [K64F GCC_ARM default] [1494585728.97][requests.packages.urllib3.connectionpool]Starting new HTTP connection (1): 54.229.57.51
10:42:10 [K64F GCC_ARM default] [1494585731.08][requests.packages.urllib3.connectionpool]http://54.229.57.51:8000 "PUT /resource/0240000034544e45002d00028aa900022011000097969900/release HTTP/1.1" 200 85
10:42:10 [K64F GCC_ARM default] [1494585731.08][requests.packages.urllib3.connectionpool]Starting new HTTP connection (1): 54.229.57.51
10:42:10 [K64F GCC_ARM default] [1494585731.18][requests.packages.urllib3.connectionpool]http://54.229.57.51:8000 "PUT /resource/0240000034544e45002d00028aa900022011000097969900/disconnect HTTP/1.1" 403 193
10:42:10 [K64F GCC_ARM default] [1494585731.18][GLRM][ERR] RemoteConnectorPrimitive.finish() failed, reason: Disconnect request failed
```

Full log:
https://jenkins-internal.mbed.com/job/ARMmbed/job/update-client-firmware-manager/job/master/37/console

Related comments:
https://jira.arm.com/browse/IOTSYST-2282

jupe, 2017-05-12:
For me this does't look like timeout issue. Logs says that disconnect fails because resource was already released. 

Anyway, PR #152 probably fix this issue.

theotherjimmy, 2017-09-19:
@marcuschangarm @jupe #152 was merged. Is this issue still present? If so, please reopen this issue.

closed 2017-09-19 by theotherjimmy

marcuschangarm, 2017-09-20:
Yes, sorry. Forgot this was still open! :smiley: