ARMmbed/htrun #172: Add user reset

geky, 2017-11-07:
I noticed there's a plugin system in place for adding additional reset logic.

It would be awfully convenient to have a reset option that simply queries the user to reset the device, and waits for the user to push a key or something.

I'm currently dealing with a board that must be power cycled for the necessary hardware blocks to initialize correctly (it's a long story). Having a user-based reset would be very convenient.

cc @studavekar, @theotherjimmy

theotherjimmy, 2017-11-07:
For automation purposes, this would stink

theotherjimmy, 2017-11-07:
I see your point though, it would be nice for new target bring up. You could develop a more robust reset method as you do other things.

geky, 2017-11-07:
For non-automation purposes it also stinks

Can't think of an alternative though 😆 

Slightly less stinky than needing to modify the tools locally

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

theotherjimmy, 2018-10-02:
@adbridge This repo does not use the same issue template as mbed-os. Your script is creating spurious errors and labels.

adbridge, 2018-10-02:
@theotherjimmy we need to fix that. All repos wanting to use the new script will also require the same issue template...

theotherjimmy, 2018-10-02:
@adbridge That has not been communicated. Also, I did not opt into using "the new script".