ARMmbed/htrun #194: QEMU plugin

gprossliner, 2018-06-14:
Hello!

How can I write a plugin to support emulated devices?

My plan would be:
1. Let green tea list and compile the tests to individual .bin files
2. I a specific test.bin file is about to run, start a QEMU session (https://gnu-mcu-eclipse.github.io/qemu/).
3. Capture the stdio from the Emulator to perform test i/o. This would be the "SerialPort".
4. If a test has completed, the qemu session is exited. There is no "reset" here, as each binary is launched in a new process.

What I basically need is to:
* Get a python method called to "flash" a binary and start the session
* Pipe stdio from the qemu session to GreenTea

How can this be accomplished?

gprossliner, 2018-06-29:
Hello! Is there any progress or information on this subject? Thanks

bridadan, 2018-07-09:
Hi @gprossliner, sorry for the very late response. For the implementation, I think you'll want to go the route that added fastmodel support: add a `ConnectorPrimitive` implementation. You can see all of the current implementations here: https://github.com/ARMmbed/htrun/tree/master/mbed_host_tests/host_tests_conn_proxy

Here's the PR that added fastmodel support, you can probably use that as a guide: https://github.com/ARMmbed/htrun/pull/187

As for getting this merged/worked on, I admit this isn't a priority for us at the moment so we'll probably only be able to support you by offering guidance and answering questions vs adding to the implementation. If you're willing to help us maintain it and test it though we'll gladly accept the contribution!

gprossliner, 2018-08-21:
Thank you, I will check the provided information.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-100*

bridadan, 2019-01-07:
Hi @gprossliner, you may have seen that htrun is moving to https://github.com/armmbed/mbed-os-tools. This week I'm going through the existing issues and doing a little clean up. Do you have any interest in continuing with this enhancement?

gprossliner, 2019-01-09:
Hi again!

Although I'm still interested in the topic, I'm currently working on a new project without mbed dependency. 
So if you are planning something in relation to this, please keep me updated.
But I will not be able to actively participate in this.

Thanks

bridadan, 2019-01-09:
Ok good to know! In that case I'll close this issue.

closed 2019-01-09 by bridadan