ARMmbed/htrun #199: Migrate Circle CI tests to Travis CI

cmonr, 2018-06-28:
https://github.com/ARMmbed/htrun/issues/196

bridadan, 2018-06-28:
@cmonr there isn't a circle ci config file in this repository, so I don't think it was running any extra tests that travis wasn't running. That's why I suggested just disabling Circle CI.

I don't have permissions, but if you look on the Circle CI config for this repo can you check to see if the test steps are specified there?

cmonr, 2018-06-29:
Ha, you're right.

Sure enough, the commands are in CircleCI settings. Coverage and pytest for the most part.

cmonr, 2018-06-29:
Pre-dependency commands:
```
pip install coverage
pip install requests[security]
pip install python-coveralls
```

Post-test commands:
```
coverage run setup.py test
coverage report
coverage html
cp -r htmlcov $CIRCLE_ARTIFACTS/coverage
```

Not sure where this is kept, but it's run before the "Post-test commands":
```
python setup.py test
```

bridadan, 2018-06-29:
> Not sure where this is kept, but it's run before the "Post-test commands":

Looks like it's (mistakenly) separating test commands and coverage commands. Travis runs the same stuff so I think you're ok closing this 👍

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-227*

closed 2019-01-07 by bridadan