ARMmbed/htrun #201: Passing cli argument to a new plugin

OmerRosler, 2018-07-05:
Hello!

I'm trying to implement a new plugin for htrun. In my plugin I need to use the cli option `-m` (indicates mcu).
However the htrun code doesn't pass this specific argument to my plugin's `execute` method.
Is this on purpose?

If it is how can I pass an additional cli arguments to my plugin?

If it isn't, would you accept a PR that fixes this?

bridadan, 2018-07-09:
Hi @OmerRosler. The reason the `-m` isn't being passed to the `execute` method is most likely because it wasn't needed. Until now! 😄 

You can add additional CLI arguments to the plugin by adding them to the `call_plugin` call for `CopyMethod` here: https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests_runner/mbed_base.py#L213

or for `ResetMethod` here: https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests_runner/mbed_base.py#L229

I would think we would accept a PR like that. Are you planning to contribute your plugin as well? And if so, can you elaborate a bit on what your plugin is doing?

OmerRosler, 2018-07-16:
We will contribute this plugin pending approval of our managers (releasing code outside is not simple for us).
It's a Jlink plugin that enables flashing and reseting over SWD or JTAG via the `jlink commander` software

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-230*

bridadan, 2019-01-07:
Hi @OmerRosler , you may have seen that htrun is moving to https://github.com/armmbed/mbed-os-tools. This week I'm going through the existing issues and doing a little clean up. Do you have any interest in continuing with this enhancement?

bridadan, 2019-01-11:
Hi @OmerRosler, we are moving the project today, so I'll be closing this issue. If this is a feature you are still interested in, please go ahead and reopen this issue! I will then transfer it to the https://github.com/armmbed/mbed-os-tools repo. Thanks!

closed 2019-01-11 by bridadan