ARMmbed/htrun #202: Softreset would trigger HTrunner to stop running subsequent tests or followup existing test case

aqib-ublox, 2018-07-05:
I want to write a watchdog test in which system should recover from Fault like Usage, Memory Management, Hard fault etc. with greentea automated test environment. 

but i am finding an issue as Fault occurs watchdog didn't get resets so continue to expire as it expires HT runner post result for display. Is there any provision in HTrunner to resume a test from place where it expires in case of soft reset so that it can be verified wdg performs recovery.  HTrunner can be modified for this use case to follow up test case after recovery from soft reset.

bridadan, 2018-07-09:
Hi @aqib-ublox, I'm afraid I'm having a bit trouble understanding the test case. But from the info you've posted it sounds a bit like the system reset tests that are already present in Mbed OS:

Device test case: https://github.com/ARMmbed/mbed-os/blob/9ddb092d4359517797999d28def84a81fa725ced/TESTS/mbed_platform/system_reset/main.cpp
Host test: https://github.com/ARMmbed/mbed-os/blob/9ddb092d4359517797999d28def84a81fa725ced/TESTS/host_tests/system_reset.py

Maybe you can use these a starting point?

aqib-ublox, 2018-07-19:
@bridadan i have explored test case and i can say yes it has helped in creating my test case thanks

aqib-ublox, 2018-07-20:
Sure closing it.

closed 2018-07-20 by aqib-ublox