ARMmbed/htrun #205: Bug in ST-LINK plugin

OmerRosler, 2018-07-16:
This relates to

https://community.st.com/thread/40136-why-did-the-behavior-of-the-st-link-cli-command-run-change

The bug in ST-LINK is not resolved to this day, despite the claims there (and that breaks the htrun plugin)

Possible workaround:

Add to `run_command` the ability to accept an `stdin` argument (defaulting to `None`), passing it to the `subprocess.call` call.
Then the ST plugin would pass some specific file handle as `stdin` (a temporary text file containing a `\n` character to workaround the bug)

Is this an acceptable solution? 
We already implemented this, would you accept it as a PR?

bridadan, 2018-07-16:
@OmerRosler Thanks for raising this. I'm sure we would accept it after some review! We don't typically use that plugin that much so we would defer to the folks that are using it. Feel free to submit the PR and we'll go from there!

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-275*

bridadan, 2019-01-07:
Hi @OmerRosler, is this still relevant? And if so, would you still like to contribute a PR?

bridadan, 2019-01-11:
Hi @OmerRosler, we are moving the project today, so I'll be closing this issue. If this is an issue you are still interested in, please go ahead and reopen this issue! I will then transfer it to the https://github.com/armmbed/mbed-os-tools repo. Thanks!

closed 2019-01-11 by bridadan