ARMmbed/htrun #211: Option to override timeout when testing app examples

MarceloSalazar, 2018-08-20:
We're using the option `--compare-log` quite a lot to test application examples and it's very useful.
https://github.com/ARMmbed/htrun#testing-mbed-os-examples

As we're testing more complex applications, we see that the default timeout (60 secs) is not enough.

Question / enhancement request: would it be possible to override this timeout with a command line option?
For example: 

```
mbedhtrun -d D: -p <com_port> -m <target> -f .\...\app.bin --compare-log compare.log --timeout 120
```

@jamesbeyond @theotherjimmy

bridadan, 2018-08-20:
Basically just need to add an option for this hardcoded value: https://github.com/ARMmbed/htrun/blob/92f5ca89e116b39dda25e73680c9c37eb9095d68/mbed_host_tests/host_tests_runner/host_test_default.py#L133

Though we already have a bunch of options for timeouts. At some point the future we may need to look into consolidating these timeouts if possible.

bridadan, 2018-08-20:
Oops, also #212 should do as well!

closed 2018-08-23 by theotherjimmy