ARMmbed/htrun #213: Can you change the read_timeout from the command line?

Alex-EEE, 2018-08-20:
Is there any way to change the timeout listed below during the handshake?

[1534794813.49][SERI][INF] serial(port=COM24, baudrate=9600, read_timeout=0.01, write_timeout=5)

bridadan, 2018-08-21:
@Alex-EEE Not at the moment, could you give specifics on the problem you're seeing?

Alex-EEE, 2018-08-21:
It's difficult to run a target in debug mode in IAR b/c of the timing with the sync packets...basically, if the unit isn't immediately ready to respond, the test fails.

Just curious, I hard coded the time to 10 s on my system and it probably won't be an issue during automated regression.

closed 2018-08-21 by Alex-EEE

bridadan, 2018-08-21:
@Alex-EEE Have you tried using the `--polling-timeout` options instead? Not sure if that will fix your issue specifically, but it'd be worth a try. If not we can add option to extend the timeout that you had to modify, just let me know!