ARMmbed/htrun #217: -e *adds* to host test list when using --list, but *overrides* mbed host tests when actually running

Alex-EEE, 2018-09-24:
Repro steps:

Use mbed test with -e to add a folder for custom host tests, and run all unit tests.  Any mbed unit test needing any of the "out of the box" host tests ( those that live in tools/host_tests ) will fail, saying it can't find tests like "RTCTest"

Although invoking mbedhtrun --list -e custom/host_tests will show the custom host tests getting added to the default ones:
+---------------+---------------------------------------------------------------------+---------------------------------------------------------------------------------------+
| name          | class                                                               | origin                                                                                |
+---------------+---------------------------------------------------------------------+---------------------------------------------------------------------------------------+
| default       | <class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>       | mbed-host-tests                                                                       |
| default_auto  | <class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>       | mbed-host-tests                                                                       |
| detect_auto   | <class 'mbed_host_tests.host_tests.detect_auto.DetectPlatformTest'> | mbed-host-tests                                                                       |
| dev_null_auto | <class 'mbed_host_tests.host_tests.dev_null_auto.DevNullTest'>      | mbed-host-tests                                                                       |
| echo          | <class 'mbed_host_tests.host_tests.echo.EchoTest'>                  | mbed-host-tests                                                                       |
| hello_auto    | <class 'mbed_host_tests.host_tests.hello_auto.HelloTest'>           | mbed-host-tests                                                                       |
| lwm2m         | <class 'lwm2m.Lwm2m'>                                               | C:\files\mbed\staging\framework\test-server-dependencies\greentea-host-tests\lwm2m.py |
| rtc_auto      | <class 'mbed_host_tests.host_tests.rtc_auto.RTCTest'>               | mbed-host-tests                                                                       |
| wait_us_auto  | <class 'mbed_host_tests.host_tests.wait_us_auto.WaitusTest'>        | mbed-host-tests                                                                       |
+---------------+---------------------------------------------------------------------+---------------------------------------------------------------------------------------+

theotherjimmy, 2018-09-28:
Huh. Thanks for the report. I'll take a look.

theotherjimmy, 2018-09-28:
I'm pretty sure that listing will add all the default host tests, while running may skip that step.

My evidence is this line https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/__init__.py#L112

@bridadan @studavekar Would it be wise to remove this line, make `-e` accept many arguments, and default to ["./tests/host_tests"]?

Alex-EEE, 2018-09-28:
I like the many folders option, b/c I can easily see a desire to pull in host tests from multiple folders

BUT, I would always just *add* those tests to the default tests, not replace them.  Most developers are probably like us:  we run all of the "out of the box" mbed tests every night on our custom board, AND, we run the unit tests we've written for our custom modules.  If you don't include the default tests, then some of those tests break.

There's really no downside to including the defaults.  They're called by name, so if you don't need them, they just sit unused.

theotherjimmy, 2018-09-28:
Hi @Alex-EEE Thanks for getting back to me. The only downside I could think of is performance (ha!).

theotherjimmy, 2018-09-28:
Further, whatever we do should be consistent between `--list` and running.

bridadan, 2018-10-01:
I'm with @Alex-EEE, we should just add the new host tests vs replace them. And we should definitely be consistent between `--list` and running.

theotherjimmy, 2018-10-01:
@Alex-EEE Maybe this is user error: try referring to the test using the name from the "name" column. Your "lwm2m" host_test happens to have a class name that's the same as the python filename without the ".py" suffix.

theotherjimmy, 2018-10-01:
BTW, the PR makes a few changes: no abspath in printing to match what's passed in, and multiple `-e` accepted.

Alex-EEE, 2018-10-01:
@theotherjimmy, do you mean referring to it inside the c code running on the DUT?  This is what my reference to the host test looks like there:  "GREENTEA_SETUP(120, "lwm2m"); "

theotherjimmy, 2018-10-02:
@Alex-EEE Yes. That matches the "name" column, which does not have a capitol "L". Perhaps you could refer to the `RTCTest` class as `rtc_auto`?

Alex-EEE, 2018-10-02:
@theotherjimmy I'm not using RTCTest, an mbed-os unit test uses that.

Just to clarify, I can get my greenTea test to pick up lwm2m when I run it by itself.  It's just an either or situation right now...I can either get all the mbed-os tests to run, but not my custom test, or I can get my custom test to run, but then all the mbed-os tests that need host tests break

theotherjimmy, 2018-10-02:
That strange. I'll have to check into how lookup works...

Alex-EEE, 2018-10-02:
Well wait, I haven't tried your PR yet...this all goes back to original problem...b/c "running" and --list don't match, it creates problems.  If you made running behave like --list, then I think everything should be golden.

theotherjimmy, 2018-10-02:
All GREENTEA_SETUP invocations in Mbed OS:
```
TESTS/mbed_drivers/c_strings/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbed_drivers/dev_null/main.cpp:    GREENTEA_SETUP(2, "dev_null_auto");
TESTS/mbed_drivers/echo/main.cpp:    GREENTEA_SETUP(30, "device_echo");
TESTS/mbed_drivers/generic_tests/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_drivers/rtc/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_drivers/stl_features/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbed_drivers/ticker/main.cpp:    GREENTEA_SETUP(test_timeout, "timing_drift_auto");
TESTS/mbed_drivers/lp_timeout/main.cpp:    GREENTEA_SETUP(240, "timing_drift_auto");
TESTS/mbed_drivers/mem_trace/main.cpp:    GREENTEA_SETUP(15, "default_auto");
TESTS/mbed_drivers/timeout/main.cpp:    GREENTEA_SETUP(240, "timing_drift_auto");
TESTS/mbed_drivers/race_test/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_drivers/flashiap/main.cpp:    GREENTEA_SETUP(120, "default_auto");
TESTS/mbed_drivers/crc/main.cpp:    GREENTEA_SETUP(15, "default_auto");
TESTS/mbed_drivers/lp_ticker/main.cpp:    GREENTEA_SETUP(test_timeout, "timing_drift_auto");
TESTS/integration/basic/main.cpp:    GREENTEA_SETUP(15, "default_auto");
TESTS/mbed_drivers/lp_timer/main.cpp:    GREENTEA_SETUP(15, "default_auto");
TESTS/mbed_drivers/sleep_lock/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_drivers/timerevent/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbed_drivers/timer/main.cpp:    GREENTEA_SETUP(15, "default_auto");
TESTS/mbedmicro-mbed/attributes/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbedmicro-mbed/call_before_main/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbedmicro-mbed/cpp/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbedmicro-mbed/div/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbedmicro-mbed/static_assert/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbedmicro-rtos-mbed/basic/main.cpp:    GREENTEA_SETUP(test_timeout, "timing_drift_auto");
TESTS/mbedmicro-rtos-mbed/mail/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbedmicro-rtos-mbed/mutex/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbedmicro-rtos-mbed/semaphore/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbedmicro-rtos-mbed/queue/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbedmicro-rtos-mbed/signals/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbedmicro-rtos-mbed/malloc/main.cpp:    GREENTEA_SETUP(test_timeout, "timing_drift_auto");
TESTS/mbedmicro-rtos-mbed/threads/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbedmicro-rtos-mbed/event_flags/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbedmicro-rtos-mbed/MemoryPool/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/events/queue/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbedmicro-rtos-mbed/condition_variable/main.cpp:    GREENTEA_SETUP(10, "default_auto");
features/frameworks/greentea-client/README.md:    GREENTEA_SETUP(5, "default_auto");
features/frameworks/greentea-client/README.md:    GREENTEA_SETUP(15, "default_auto");
TESTS/mbedmicro-rtos-mbed/heap_and_stack/main.cpp:    GREENTEA_SETUP(test_timeout, "default_auto");
TESTS/mbedmicro-rtos-mbed/rtostimer/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/events/timing/main.cpp:    GREENTEA_SETUP((number_of_cases + 1)*TEST_EVENTS_TIMING_TIME / 1000, "default_auto");
TESTS/mbedmicro-rtos-mbed/systimer/main.cpp:    GREENTEA_SETUP(5, "default_auto");
TESTS/mbedmicro-rtos-mbed/kernel_tick_count/main.cpp:    GREENTEA_SETUP(10, "timing_drift_auto");
features/frameworks/utest/README.md:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_functional/callback/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbed_functional/callback_big/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbed_functional/callback_small/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbed_functional/functionpointer/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbed_hal/lp_ticker/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_hal/rtc_time/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_hal/sleep_manager/main.cpp:    GREENTEA_SETUP(20, "default_auto");
TESTS/mbed_hal/sleep_manager_racecondition/main.cpp:    GREENTEA_SETUP(30, "default_auto");
TESTS/mbed_hal/critical_section/main.cpp:    GREENTEA_SETUP(10, "timing_drift_auto");
TESTS/mbed_hal/ticker/main.cpp:    GREENTEA_SETUP(60, "default_auto");
TESTS/mbed_hal/rtc_time_conv/main.cpp:    GREENTEA_SETUP(300, "rtc_calc_auto");
TESTS/mbedtls/selftest/main.cpp:    GREENTEA_SETUP(120, "default_auto");
TESTS/mbed_hal/common_tickers/main.cpp:    GREENTEA_SETUP(80, "default_auto");
TESTS/mbedtls/multi/main.cpp:    GREENTEA_SETUP(10, "default_auto");
TESTS/mbed_hal/common_tickers_freq/main.cpp:    GREENTEA_SETUP(120, "timing_drift_auto");
TESTS/mbed_hal/crc/main.cpp:    GREENTEA_SETUP(30, "default_auto");
TESTS/mbed_hal/qspi/main.cpp:    GREENTEA_SETUP(180, "default_auto");
TESTS/mbed_hal/rtc/main.cpp:    GREENTEA_SETUP(60, "default_auto");
TESTS/mbed_hal/rtc_reset/main.cpp:    GREENTEA_SETUP(100, "rtc_reset");
TESTS/network/wifi/main.cpp:    GREENTEA_SETUP(240, "default_auto");
TESTS/mbed_hal/sleep/main.cpp:    GREENTEA_SETUP(60, "default_auto");
TESTS/network/interface/main.cpp:    GREENTEA_SETUP(480, "default_auto");
TESTS/mbed_hal/us_ticker/main.cpp:    GREENTEA_SETUP(20, "default_auto");
[... cut ...]
components/storage/blockdevice/COMPONENT_SD/TESTS/filesystem/fopen/fopen.cpp:    GREENTEA_SETUP(FSFAT_FOPEN_GREENTEA_TIMEOUT_S, "default_auto");
components/storage/blockdevice/COMPONENT_FLASHIAP/TESTS/filesystem/fopen/fopen.cpp:    GREENTEA_SETUP(FSLITTLE_FOPEN_GREENTEA_TIMEOUT_S, "default_auto");
components/storage/blockdevice/COMPONENT_SD/TESTS/filesystem/seek/main.cpp:    GREENTEA_SETUP(MBED_TEST_TIMEOUT, "default_auto");
components/storage/blockdevice/COMPONENT_SD/TESTS/filesystem/parallel/main.cpp:    GREENTEA_SETUP(MBED_TEST_TIMEOUT, "default_auto");
```

None invoke it with `"RTCTest"`, and all use the "name" column name.

theotherjimmy, 2018-10-02:
I think run and list use the same registry, the global `mbed_host_tests.HOSTREGISTRY`.

Alex-EEE, 2018-10-02:
Ok, so RTCTest was just a guess, I could remember from my output exactly which host tests it was looking for, just that it couldn't find it anymore.

Right now my observation is that if you supply -e, then --list will include all tests in mbed_host_tests.HOSTREGISTRY in addition to tests supplied with -e.  But run will not...for running, it appears -e *overrides* mbed_host_tests.HOSTREGISTRY

If you can add a custom host test somehow in a way I haven't tried, and you can get greentea to pick it up, let me know what you did and by all means I'll do it that way.  But every different way I've tried using -e, I lose the tests inside mbed_host_tests.HOSTREGISTRY

theotherjimmy, 2018-10-02:
@Alex-EEE I can find nothing in the code that would make the new tests override old ones, nor could I find any use of the `unregester_host_test` function, the only way to remove host tests.

It looks like the only way that host tests are looked up is through `get_host_test`..

-------------

I think this is a windows-only issue: HOSTREGISTRY may be empty on the new Process's side of things. I ran into this sort of thing before (in particular https://github.com/ARMmbed/mbed-os/pull/7477). We should probably not have a global var anyway.

Alex-EEE, 2018-10-02:
Ahhhh, the windows thing about HOSTREGISTRY makes a lot of sense.  https://github.com/ARMmbed/htrun/blob/63eea4ee94a88f8e5c292588889fc243b5177aee/mbed_host_tests/__init__.py#L166  does appear to just add tests, and I remember when I saw that, I was confused why the original tests were going away

theotherjimmy, 2018-10-02:
Well this means that the registry needs to not be a global, and that those "helper functions" need to go away.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-493*

PR #219 (2018-11-05): WIP: make `--list` conistant with running WRT `-e`
Resolves #217
closed 2018-11-05 by theotherjimmy