ARMmbed/htrun #222: htrun fails with mbedgt: cannot import name MbedBoard

studavekar, 2018-10-08:
steps to reproduce 

- create a fresh virtualenv 
- install htrun ` pip install mbed-host-tests`

run `mbedgt`
```
(test_htrun_gt_update) ➜  mbed-os mbedgt --help
mbedgt: Not all required Python modules were imported!
mbedgt: cannot import name MbedBoard
mbedgt: Check if:
        1. You've correctly installed dependency module using setup tools or pip:
                * python setup.py install
                * pip install <module-name>
        2. There are no errors preventing import in dependency modules
        See: https://github.com/ARMmbed/greentea#installing-greentea
```

This **effectivley breaks CI system** which would create a virtualenv and install packages using pip.

The root cause looks like latest release of [pyOCD](https://github.com/mbedmicro/pyOCD/releases/tag/v0.13.0b1)  with connector class ``` New ConnectHelper class with static methods previously in MbedBoard.```

SenRamakri, 2018-10-08:
Adding @theotherjimmy and @bridadan

studavekar, 2018-10-23:
closing it as fixed is been merged.

closed 2018-10-23 by studavekar