ARMmbed/htrun #225: How can i implement htrun (Key-value) protocol without using greentea or greentea client, or mbed-ls?

mukundsharma1995, 2018-10-17:
I am following this GitHub link for htrun -- https://github.com/ARMmbed/htrun
But what I want to do is to implement Htrun on host side, The DUT side will be taken care of, Is there any way to use the functionality of HTRUN on host and exchanges key-value messages between DUT and host.

The above resource explains the interaction between greentea client and htrun, I didn't find any specific example for fully implement htrun and exchange key;values

What i tried is mbedhtrun to flash and reset, after that?? how can I send data serially and get response, not able to find specific resource for this.

I am new to this mbed and htrun please share some useful resources.

what I tried looks like:

-- mbedhtrun -d D: -p /dev/ttyUSB0:115200
-- {{__sync;5f8dbbd2-199a-449c-b286-343a57da7a37}}

after this how can I use those setup(), and callback methods?
how can i use serial Tx and Rx to communicate from raspberry pi to DUT?

I am not able to completely understand greentea and htrun, a good example will be appreciated..
Thanks in advance

@ARMmbed
@bridadan

mukundsharma1995, 2018-10-17:
yes, my aim is to implement both DUT side and HOST side, but not able to understand this greentea framework concept , It is like forcing all the steps, what if i dont want to start execution with __sync?, i have tried implementing htrun independently but didn't work for me. as i have to use greentea client api to use htrun which i don't want

bridadan, 2019-01-07:
Hi @mukundsharma1995 , you may have seen that htrun is moving to https://github.com/armmbed/mbed-os-tools. This week I'm going through the existing issues and doing a little clean up. Do you have any interest in pursuing this further?

bridadan, 2019-01-11:
Hi @mukundsharma1995 , we are moving the project today, so I'll be closing this issue. If this is an issue you are still interested in, please go ahead and reopen this issue! I will then transfer it to the https://github.com/armmbed/mbed-os-tools repo. Thanks!

closed 2019-01-11 by bridadan