ARMmbed/htrun #227: Error on running htrun with stlink plugin

masoudr, 2018-11-08:
Hi,
I'm trying to use stlink plugin to upload a simple example but I get this error:

> mbedhtrun -f ./mbed-example.bin -c stlink -r stlink
> [1541667888.67][HTST][INF] host test executor ver. 1.4.2
> [1541667888.68][HTST][INF] copy image onto target...
> Traceback (most recent call last):
>   File "C:\Python27\Scripts\mbedhtrun-script.py", line 11, in <module>
>     load_entry_point('mbed-host-tests==1.4.2', 'console_scripts', 'mbedhtrun')()
>   File "c:\python27\lib\site-packages\mbed_host_tests\mbedhtrun.py", line 34, in main
>     result = test_selector.execute()
>   File "c:\python27\lib\site-packages\mbed_host_tests\host_tests_runner\host_test_default.py", line 524, in execute
>     result = self.mbed.copy_image()
>   File "c:\python27\lib\site-packages\mbed_host_tests\host_tests_runner\mbed_base.py", line 187, in copy_image
>     initial_remount_count = get_remount_count(disk)
>   File "c:\python27\lib\site-packages\mbed_host_tests\host_tests_runner\mbed_base.py", line 97, in get_remount_count
>     files_on_disk = [x.upper() for x in os.listdir(disk_path)]
> TypeError: coercing to Unicode: need string or buffer, NoneType found
> 

I'm using htrun v1.4.2.

0xc0170, 2018-11-13:
What is your stlink version (target that you are testing) ?

cc @ARMmbed/team-st-mcd

masoudr, 2018-11-13:
I'm using ST-link original model v2.0 on Windows. My target is an STM32F407.

bridadan, 2019-01-07:
Hi @masoudr , you may have seen that htrun is moving to https://github.com/armmbed/mbed-os-tools. This week I'm going through the existing issues and doing a little clean up. Is this still an issue?

bridadan, 2019-01-11:
Hi @masoudr, we are moving the project today, so I'll be closing this issue. If this is an issue you are still interested in, please go ahead and reopen this issue! I will then transfer it to the https://github.com/armmbed/mbed-os-tools repo. Thanks!

closed 2019-01-11 by bridadan